AndrewEllis93/Ellis-SuperSlicer-Profiles #3: Internal bridging issue on z_drive_main parts

Jasrags, 2021-10-30:
First off thanks for the profiles, they have been great in helping me understand the more advanced tuning i needed to do to push my profiles forward.

I just ran into a strange issue using your standard profile that i wanted to share and see if it's a problem on your side as well. 

Steps to reproduce:

1. Using SS .57.0 or .57.3
1. Load and slice `z_drive_main_b_x2.stl`

[image]
[image]
[image]

What seems to be happening is due to the `180%` infill extrusion width the slicer is unable to complete the infill and decides to use `Internal Bridge Infill`. 

Fix:
Reduce `Internal Bridge Infill` <= `160%`

I'm in no way saying this is not a super slicer issue, but i wanted to raise it to you as well to see if you have it on your prints or you know of any other way to solve for this. 

**Note**: Some modifications have been made during the tuning of my printer but nothing in the width section.

3MF file: [z_drive_main_a_x2.3mf.zip](https://github.com/AndrewEllis93/Ellis-PIF-Profile/files/7445426/z_drive_main_a_x2.3mf.zip)

AndrewEllis93, 2021-10-30:
I'm not quite sure I understand what the issue is. Can you circle the issue in your example photos? Do you mean that little corner in one of the two cutouts with the bridging?
And I don't see a setting called "internal bridge infill", though I do see it as an extrusion type in the preview.

Jasrags, 2021-10-30:
Hey, sorry i should have done that at first. It happens at layer 17 (3.45mm z). 

[image]

Jasrags, 2021-10-30:
There is no setting for internal bridge infill as far as i can see, i think it's just one of those strange slicer issues with line widths but i just wanted to raise it to you as it might be happening for others with these settings.

Jasrags, 2021-10-30:
Also i see another typo the width is `Infill: width` not what i said before.

AndrewEllis93, 2021-11-19:
This appears to have been resolved with today's change / the new SS version:

2021-11-19: Set new bridge_overlap_min setting to 50%
- Fixes 50% bridge density in new SuperSlicer version (2.3.57.6).

closed 2022-01-05 by AndrewEllis93