AndrewEllis93/Ellis-SuperSlicer-Profiles #7: Importing PIF Profile (45 degree) in 2.4.58.2 Has Wall Thickness Bug

adawalli, 2022-04-13:
Hey there - was just checking out your latest PIF profile, and I see the following issue when importing your latest PIF profile for 45 degree. This parameter in particular (`s_wall_thickness`) doesn't actually appear to be in your .ini file (which might be why SS is being so weird)
[image]

AndrewEllis93, 2022-04-13:
I think this is just a bug in SS. If I change tabs back and forth, the value keeps hopping around. Tried saving it as 0, and tried adding it to the .ini file as 0 (which is supposed to be the default), but no difference.

adawalli, 2022-04-13:
ok - will report on SS issues - thanks

adawalli, 2022-04-13:
Closing as this isn't really a profile issue - however, you may want to warn your users @AndrewEllis93 in the Readme...I have no idea what these random values will do to prints (at one point, it picked 36856 as a value)

closed 2022-04-13 by adawalli

supermerill, 2022-04-13:
It's a quick setting I added for the simple mode, still a bit work in progress.
just forgot to convert the mode from "expert" to "Expert", so it's currently starting with no mode, hence why you're seeing too many fields. Click on "expert" and you should see it disappear.