AndrewEllis93/Ellis-SuperSlicer-Profiles #8: Partially preheat hotend while QGL runs

docwelch, 2022-04-20:
Not really an issue but wasn't sure it warranted a pull request either. I agree with you it's nice to not have filament oozing while the QGL is going on but I hate to wait so long for the hotend to get up to temp. So, my solution is to partially preheat while QGL is occurring. I do this by adding another variable to the custom Start G-code section in Printer Settings in SuperSlicer:

`PREHEATHOTEND={(first_layer_temperature[initial_extruder]+extruder_temperature_offset[initial_extruder])*.70}`

And then, in my printer.cfg file (PRINT_START macro), I use:

`{% set preheathotendtemp = params.PREHEATHOTEND|default(150)|int %}`

And then:

`M104 S(preheathotendtemp)`

right after heating the bed and before G32.

I've been preheating to 70% of the printing temp and do not see oozing. You might be able to go even higher but I have not tried.

AndrewEllis93, 2022-04-20:
My PRINT_START macro actually does the same (or a similar) thing - just one of those things left up to the user, since each person's preferred PRINT_START varies.

If this were to be in the guide, it would be best added as a note [here](https://github.com/AndrewEllis93/Print-Tuning-Guide/blob/main/articles/passing_slicer_variables.md) or something, but I'm not sure.

AndrewEllis93, 2022-04-20:
If you want to PR it to that guide - could add a "Tips and Tricks" section at the bottom or something like that, rather than being in the primary config example

closed 2022-04-20 by AndrewEllis93

viniavila, 2024-10-01:
I fixed in 150°C the preheating temperature for HOMING, QGL and BED_MESH. But I put nothing in SuperSlicer, I opted to leave this be managed by my PRINT_START macro. I only pass the bed and nozzle temperature to it (my code start code is just PRINT_START BED=<XXX> EXTRUDER=<YYY> in SS, leaving the details in printer.cfg)
`[gcode_macro PRINT_START]
gcode:  
    RESPOND MSG="AQUECENDO A MESA E O BICO"
    M117 Heating Bed and Nozzle
    HEATSOAK
    M190 S{params.BED}
    M109 S150

    RESPOND MSG="FAZENDO HOMING E QGL"
    M117 Homing e QGL

    {% if "xyz" not in printer.toolhead.homed_axes %}
      SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=homing VALUE=True
      STATUS_HOMING
      G28
      SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=homing VALUE=False
    {% endif %}
    
    {% if printer.quad_gantry_level.applied == False %}
      SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=qgling VALUE=True
      STATUS_MESHING
      QGL
      G28
      SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=qgling VALUE=False
    {% endif %}
    
    RESPOND MSG="FAZENDO BED MESH CALIBRATE"
    M117 Calibrating Mesh
    SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=probing VALUE=True
    STATUS_LEVELING
    BED_MESH_CALIBRATE ADAPTIVE=1
    SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=probing VALUE=False
    
    G90                            ; absolute positioning
    
    RESPOND MSG="POSICIONANDO A TOOLHEAD PERTO DA ÁREA DE IMPRESSÃO"
    M117 Smart parking
    SMART_PARK
    
    RESPOND MSG="AQUECENDO O BICO ATÉ A TEMPERATURA DE IMPRESSÃO"
    M117 Heating Nozzle
    STATUS_HEATING
    M104 S{params.EXTRUDER}
    M109 S{params.EXTRUDER}
    
    RESPOND MSG="PURGANDO O BICO DE IMPRESSÃO"
    M117 Purging Filament
    STATUS_CLEANING
    LINE_PURGE

    SFS_ENABLE    ; enable filament sensor
    M117
    STATUS_PRINTING`