AndrewEllis93/Ellis-SuperSlicer-Profiles #10: volumetric speed

supermerill, 2022-07-21:
Hi
I have detected a little error
when you say

> The volumetric speed limit controls the maximum rate that your hotend is allowed to extrude.
>
>No matter how much you push speeds, layer heights, or line widths, it will never allow you to outrun your hotend.

But in the picture, it's the (max) flow for the auto-speed feature (when a speed is set to 0). 
The universal flow limit you talk about is in the filament tab.

AndrewEllis93, 2022-07-25:
My understanding was that the flow in the filament tab just overrides that setting if a particular filament acts differently. 

When I change the number in the print settings, it does seem to apply 

[image]
[image]

supermerill, 2022-07-25:
Dang, you're right.
I modified it a year ago and forgot to save it in my brain.
I'll modify the tooltip to reflect the behavior.

closed 2022-07-25 by supermerill

dewi-ny-je, 2022-07-27:
And yet it's likely a good idea to tell people to put that limitation in the filament tab: PLA and ABS have quite different flow limits for a given extruder/hotend.

Also, maybe a couple of tips about different nozzles: I use different filament settings ("PLA 0.6", "PLA 0.4") so that each one can have a different flow limit. 
Obviously I also have multiple "printer" profiles for the nozzles.