AndrewEllis93/Ellis-SuperSlicer-Profiles #11: Practical tips about accelerations

dewi-ny-je, 2022-07-27:
You write 

```
This profile's speeds/accels are tuned for linear rail CoreXY (V2/V1/Trident/V0). For other printer types (Switchwire, Legacy, others), you will likely need to turn down some speeds and accelerations.

I actually use the same print settings on my Ender 3, just with speeds and accelerations toned down
```

but it's not so clear how much to reduce/increase over the IS suggestions.
You could add a line about your printer: "IS suggests 5000 k for my printer, so I picked xx% for external perimeters, yyy% for sparse infill and zzz% for travel (assuming the motors have enough current to allow it)".

I didn't check the ratios, I just threw there some placeholders.

AndrewEllis93, 2022-09-12:
Unfortunately it's not as simple as rules of thumb, it just varies a ton depending on components, build quality, design, etc. 

I'll mull over how to potentially word it a bit better.

EricZimmerman, 2022-10-21:
perhaps posting the pngs for your input shaper x and y would allow for comparisons to what others look like?

LAP87, 2022-11-22:
In your profiles, is it possible that you could start using the percentages in the acceleration fields? And then we just input our shaper result as the default/standard?

AndrewEllis93, 2022-11-24:
@LAP87 In my experience, across similar printers, the input shaper result is more of a ceiling than something that linearly scales all your accels. At least with how I personally handle accels (keep them low where you can see it, go nuts where you can't). I think scaling it with the max doesn't really align with that