AndrewEllis93/Ellis-SuperSlicer-Profiles #15: Unable to find wall_thickness_lines in SuperSlicer

lincolnep, 2022-11-09:
Was updating my SuperSlicer profiles to the latest version by just following a long with the changelog.

But I am unable to find wall_thickness_lines so I have had a look inside the ini profile and found that it is a print_custom_variables = wall_thickness_lines = 0 

Is that correct?

Also I am not sure what that is used by would you be able to explain what it does please.

Thank you for so much time and effort into these profile I very much appreciate the work you put into them.

Lincoln

AndrewEllis93, 2022-11-09:
There's something really odd about that setting. When you save it, it saves to custom variables. I don't know why it's the only settings to do that.

AndrewEllis93, 2022-11-09:
Added a note to change log about this in 314c2a40170180180c80aed0d831bde58264b5ea.

closed 2022-11-09 by AndrewEllis93