AndrewEllis93/Ellis-SuperSlicer-Profiles #16: Unable to load configuration.

skyaee, 2022-11-11:
Unable to load configuration.
[image]
[image]
[image]

AndrewEllis93, 2022-11-11:
Load as config, not as config **bundle**.

Also please try the recommended version of SuperSlicer (2.4.58.5).

skyaee, 2022-11-11:
> 导入为配置，而不是配置**包**。
> 
> 也请尝试推荐版本的 SuperSlicer (2.4.58.5)。
> 
> 你是按照说明下载的吗？
[image]
Yes, follow the tutorial process exactly, the same is true for version 2.4.58.

skyaee, 2022-11-11:
https://user-images.githubusercontent.com/40803751/201381953-2c6001a8-6822-489b-9cc5-ff17ff83df6c.mov

AndrewEllis93, 2022-11-11:
I am unable to replicate this issue. 
Please upload the exact ini file you are using here, just want to make sure it looks good.
Could also be something different with the Mac version, I use Windows

skyaee, 2022-11-11:
[config.ini.zip](https://github.com/AndrewEllis93/Ellis-SuperSlicer-Profiles/files/9992285/config.ini.zip)
My current configuration file is the default configuration file of the software.

AndrewEllis93, 2022-11-11:
Sorry, I mean the ini file you are trying to import, in your video

skyaee, 2022-11-11:
[归档.zip](https://github.com/AndrewEllis93/Ellis-SuperSlicer-Profiles/files/9992312/default.zip)
Sorry, I misunderstood, this is the file.

AndrewEllis93, 2022-11-11:
You have downloaded the file as HTML instead of raw
Are you using the "raw" button as shown?
[image]

skyaee, 2022-11-11:
Thank you, it is the file download problem, I have no problem with the new download. thanks.

closed 2022-11-11 by AndrewEllis93