AndrewEllis93/Ellis-SuperSlicer-Profiles #17: logic behind extrusion widths?

firsttest12, 2022-11-20:
Eager to understand the logic behind the use of 

 - 1x (100%) extrusion width for "normal" parts: 0.4mm for a 0.4mm nozzle etc. ?

- 115% extrusion width for the perimeters of Decorative parts?

Simply because:

- it's commonly argued that, in order to get better binding between perimeters, it's better to have extrusion widths around 120%: because at 100% the extrudate is significiantly rounded (beause it has come out of a round aperture and has not been squished under that flat portion of the nozzle) at the edges (also layer height dependent). Would have expected that to be used for the "normal" (functional) profiles.

Additionally I've sort of noticed that extruding at 100% (1x) can give parts with a "cleaner" finish, that in some ways are therefore more aesthetic. ( For example prints seem to show less Z banding, perhaps because minor Z fluctuations are "absorbed" by the slightly curved edges of the outermost perimeter.) Might have expected 100% to be use for aesthetic parts.

In other words, genuinely confused.

AndrewEllis93, 2022-11-20:
The PIF profile only has 100% external perims because it's a requirement for Voron parts - I don't actually like 100% widths, and took the opportunity to fix it in my decorative profiles, partially in case someone is using variable layer heights and exceeds 50% layer heights.

firsttest12, 2022-11-20:
Thanks.

Voron's 1x logic seems to be lost in the mists of time. Odd.

In the meantime, thanks for the excellent - and thought-provoking - profiles!

AndrewEllis93, 2022-11-20:
You're welcome!

closed 2022-11-20 by AndrewEllis93