AndrewEllis93/Ellis-SuperSlicer-Profiles #25: Need profiles that are Arachne compatible

jwhite, 2024-04-23:
Some of the profiles could/should be updated to support the arachne slicer, or at least some guidance given on good parameters.

Are these profiles still being maintained?

morlan, 2025-02-17:
These profiles have helped me tremendously get started printing ABS with my Voron 2.4. Thanks for providing them.

I'm also curious if these profiles have/will be updated to the latest version of SuperSlicer or ported to other slicers i.e. OrcaSlicer. Thanks again for your great tuning guide and all the resources you have provided to the community!

bassamanator, 2025-04-19:
> Some of the profiles could/should be updated to support the arachne slicer, or at least some guidance given on good parameters.
> 
> Are these profiles still being maintained?

SuperSlicer just had an update roughly 2 weeks ago, so in fact it hasn't been possible to update.